ethz-adrl/ifopt #7: Issues running catkin_make

rshum19, 2018-02-10:
Hi, 

I have updated the CMakeList.txt file to point to the location of my Ipopt installation. Ipopt is working correctly since I have used it with other programs. When I try to run the 'catkin_make' command if get the following error:

`[ 75%] Built target ifopt_core
make[2]: *** No rule to make target /usr0/home/rshu/libraries/CoinIpopt/build/lib/libipopt.so, needed by /usr0/home/rshu/catkin_ws_sandbox2/devel/lib/libifopt_ipopt.so.  Stop.
make[1]: *** [ifopt/ifopt_ipopt/CMakeFiles/ifopt_ipopt.dir/all] Error 2
make: *** [all] Error 2`

any suggestions why this could be?

Thank you,
Roberto

awinkler, 2018-02-10:
Can you verify that libipopt.so is actually in the folder /usr0/home/rshu/libraries/CoinIpopt/build/lib/
If not you might have to run "make install" in ipopt after running "make". This should put the shared libraries in the folder and then allow catkin to find them. Give that a try.

rshum19, 2018-02-10:
Thank you for the prompt response. I found the issue. When installing Ipopt I disabled shared library compilation to make the libraries Matlab compatible. Thus, I had to change in the CMakeList.txt file the line

'set(IPOPT_LIBRARIES    ${IPOPT_DIR}/build/lib/libipopt.so) '

to 

set(IPOPT_LIBRARIES    ${IPOPT_DIR}/build/lib/libipopt.a) 

With that change I managed to compile ifopt without issues.

closed 2018-02-10 by rshum19