ethz-adrl/ifopt #8: catkin_make run_tests failure

lplp8899, 2018-02-19:
hey Alex, I can run catkin_make. but when I run catkin_make run_testsm, I got the following error:

_Exception of type: OPTION_INVALID in file "IpAlgBuilder.cpp" at line 345:
 Exception message: Selected linear solver MA57 not available.
Tried to obtain MA57 from shared library "libhsl.so", but the following error occured:
libhsl.so: cannot open shared object file: No such file or directory

EXIT: Invalid option encountered.
Ipopt failed to find a solution. ReturnCode: -120.5 1.5_

I guess that is because my library for MA57 is called libcoinhsl.so which is located in /home/penglu/packages/Ipopt/ThirdParty/HSL/coinhsl/lib, could you please give me a hint how to solve this issue? Thanks.

awinkler, 2018-02-20:
Hi!
Yes, you are missing the solver MA57. So you have two options:

1. Download and install _Coin-HSL Full_ from here http://www.hsl.rl.ac.uk/ipopt/ , this will allow you to use this solver.
2. Just use a solver that you have installed, probably MA27, by changing the parameter in this source file: https://github.com/ethz-adrl/ifopt/blob/9fca0a7be8d6b889f768fe97217fc09abd733d86/ifopt_ipopt/src/ipopt_adapter.cc#L66

lplp8899, 2018-02-20:
thanks Alex!

closed 2018-02-21 by awinkler