ethz-adrl/ifopt #10: How to use Eigen as solver library

jbensabat, 2018-04-05:
Hello
I have compiled IFOPT and linked to IOPT. In the IFOPT site it is specified that it interfaces with EIGEN. Does this means that EIGEN can be used as a linear solver in IPOPT or just that EIGEN is used for the formulation of the optimization problem. 
thank you for clarifying. 
jacob

awinkler, 2018-04-05:
Hi Jacob,

Yes, the idea is that you can use Eigen vectors, matrices and the efficient arithmetic operations provided by Eigen to calculate the costs and constraints; so Eigen for the higher-level problem specification. 

The linear solvers IPOPT uses internally are not provided by IFOPT. However, you can specify these solvers here:
https://github.com/ethz-adrl/ifopt/blob/36234090e1d4cc31ecbdc5fc40eab7edd6508687/ifopt_ipopt/src/ipopt_adapter.cc#L66

Best
Alex

jbensabat, 2018-04-05:
Hi
Thanks for your answer
Understood
Best
Jac

Jacob Bensabat, PH.D.
CTO
EWRE Ltd.
Tel: +972-4-8383919.
Fax: +972-4-8387621.
Mobile: +972-544-417511.

https://github.com/ethz-adrl/ifopt/blob/36234090e1d4cc31ecbdc5fc40eab7edd6508687/ifopt_ipopt/src/ipopt_adapter.cc#L66

closed 2018-04-05 by awinkler