ethz-adrl/ifopt #12: Difficulties with shared libraries and improved ifopt_snopt CMake

A-I-Ivanov, 2018-04-17:
Hello, 
We intend to use your nice wrapper for SNOPT 7.6. We've had some issues linking libraries. When compiling ifopt the following error message occurs when linking.

` Linking CXX shared library /home/alex/gitprojects/untrusted_maps_planner/catkin_ws/devel/lib/libifopt_snopt.so
/usr/bin/ld: /usr/local/lib/libsnopt7_cpp.a(snoptProblem.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libsnopt7_cpp.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
make[2]: *** [/home/alex/gitprojects/untrusted_maps_planner/catkin_ws/devel/lib/libifopt_snopt.so] Error 1
make[1]: *** [ifopt/ifopt_snopt/CMakeFiles/ifopt_snopt.dir/all] Error 2 `

We resolved this my modifying lines 54 and 56 of the CMakeLists file for ifopt_snopt to be as follows.

`if(${SNOPT76})
 add_library(${PROJECT_NAME} STATIC src/snopt76_adapter.cc)
else()
 add_library(${PROJECT_NAME} STATIC src/snopt_adapter.cc)
endif()`

This work around is ok, but I'm wondering if we can actually use shared libraries and how.

Finally, we also modified line 15 (same file) to be in-line with the SNOPT installation instructions. I think this may help make the code a bit more modular between machines. 

'set(SNOPT_DIR "/home/$ENV{USER}/software/snopt_lib")'

awinkler, 2018-04-18:
Hi alex,

Thanks for raising this issue. Can you please try to use
`add_compile_options(-std=c++11 -fPIC)`
here:
https://github.com/ethz-adrl/ifopt/blob/fbf7acda4e3e42711031f65e015f6c9f84c87fbd/ifopt_snopt/CMakeLists.txt#L4
and removing the STATIC keywords you added below in the file. 

Please let me know if this works. 
Best
Alex

A-I-Ivanov, 2018-04-20:
I attempted that fix, but the problem persists.  Here is the output:

'''
Linking CXX shared library /home/alex/gitprojects/untrusted_maps_planner/catkin_ws/devel/lib/libifopt_snopt.so
/usr/bin/ld: /home/alex/software/snopt_lib/lib/libsnopt7_cpp.a(snoptProblem.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/home/alex/software/snopt_lib/lib/libsnopt7_cpp.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
make[2]: *** [/home/alex/gitprojects/untrusted_maps_planner/catkin_ws/devel/lib/libifopt_snopt.so] Error 1
make[1]: *** [ifopt/ifopt_snopt/CMakeFiles/ifopt_snopt.dir/all] Error 2
make: *** [all] Error 2
*** Failure: Exit code 2 ***

'''

awinkler, 2018-06-25:
Hi @A-I-Ivanov 
I just added quite some improvements (see #17) and would be curious if this fixes your issue. If you find time, could you kindly let me know? Thanks :)

A-I-Ivanov, 2018-06-25:
Hi Alex,
I may give it a go today. I'm having other compiler issues, so this is in
the same vein. Thanks for taking the time.

Cheers,
A.I.

closed 2018-08-08 by awinkler