ethz-adrl/ifopt #13: Help Installing Ifopt-Ipopt w/o catkin label:"help wanted"

shoshievass, 2018-04-24:
I'm hoping to build Ifopt (with ipopt in particular) on a Mac-OSX without the use of catkin.

I have Ipopt and Eigen installed properly, but I'm not sure about which lines need to be added/changed for a standard `mkdir build; cd build; cmake ..` to work. 

In particular, I've put in my Ipopt directory in the `set(IPOPT_DIR ... )` line of [CMakeLists.txt](https://github.com/ethz-adrl/ifopt/blob/fbf7acda4e3e42711031f65e015f6c9f84c87fbd/ifopt_ipopt/CMakeLists.txt#L16-L17) but I'm not totally sure what it means to ensure that "IPOPT headers and library are installed
# somewhere in the default search paths", and I got the following error:

>   Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

Apologies for the novice question but I would very much appreciate some assistance!

jwdinius, 2018-06-12:
@shoshievass I have a similar need, so I'll have a crack at this tomorrow.

shoshievass, 2018-06-12:
Awesome! Please let me know how it goes!

jwdinius, 2018-06-13:
I will submit a pr that you can look at in the next few days.

awinkler, 2018-06-14:
Great, thanks @jwdinius ! 
I'll have a look at it in the next few days.

jwdinius, 2018-06-16:
I'm changing the pr to work-in-progress so you all can have a look at it and make comments.  I would like to clean up the implementation a little more before it's merged.

awinkler, 2018-06-25:
Hi @jwdinius, @shoshievass 
Building with pure cmake works now (see #17), please let me know if you experience any problems.

Thanks for your help on this @jwdinius, sorry I couldn't merge your PR, I decided to do some beneficial structural code improvements as well while adding the cmake features.

jwdinius, 2018-06-26:
No worries.  I actually thought that I had updated the PR with other work I had done, but I guess I was mistaken.  For your reference, https://github.com/jwdinius/ifopt/tree/pure_cmake.

I followed the outline provided by kindr in most places, including setting up package exporting.  Now, ifopt_ipopt(snopt) can use `find_package()` to find `ifopt_core` during the local superbuild or via an external cmake project.  Please have a look if you are interested in exporting these projects, which might be handy when trying to install ifopt from within another project (like how kindr is built within https://bitbucket.org/adrlab/ct).  

I was able to verify that the packages were able to found from an external cmake project built with unit tests.  Hopefully something in my repo will be of use to you.  Cheers!

sapan-ostic, 2018-07-13:
Hi,
I am trying to build using CMake. I have made changes according to instructions in the CmakeLists.txt. But I am getting the following error.

**~/MyProject$ ls**
bin       build      CMakeLists.txt  example.cpp  libipopt2.so

**$cmake --build build -- -j3**
[ 50%] Linking CXX executable ../bin/example
//usr/local/lib/libipopt.so.1: undefined reference to `dpotrf_'
//usr/local/lib/libipopt.so.1: undefined reference to `dgemm_'
//usr/local/lib/libipopt.so.1: undefined reference to `dnrm2_'
//usr/local/lib/libipopt.so.1: undefined reference to `dscal_'
//usr/local/lib/libipopt.so.1: undefined reference to `dpotrs_'
//usr/local/lib/libipopt.so.1: undefined reference to `dtrsm_'
//usr/local/lib/libipopt.so.1: undefined reference to `dlopen'
//usr/local/lib/libipopt.so.1: undefined reference to `daxpy_'
//usr/local/lib/libipopt.so.1: undefined reference to `dgetrf_'
//usr/local/lib/libipopt.so.1: undefined reference to `dcopy_'
//usr/local/lib/libipopt.so.1: undefined reference to `idamax_'
//usr/local/lib/libipopt.so.1: undefined reference to `dlclose'
//usr/local/lib/libipopt.so.1: undefined reference to `dlerror'
//usr/local/lib/libipopt.so.1: undefined reference to `dlsym'
//usr/local/lib/libipopt.so.1: undefined reference to `dsyrk_'
//usr/local/lib/libipopt.so.1: undefined reference to `dasum_'
//usr/local/lib/libipopt.so.1: undefined reference to `dsyev_'
//usr/local/lib/libipopt.so.1: undefined reference to `dmumps_c'
//usr/local/lib/libipopt.so.1: undefined reference to `dgetrs_'
//usr/local/lib/libipopt.so.1: undefined reference to `ddot_'

collect2: error: ld returned 1 exit status
CMakeFiles/example.dir/build.make:96: recipe for target '../bin/example' failed
make[2]: *** [../bin/example] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/example.dir/all' failed
make[1]: *** [CMakeFiles/example.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

jwdinius, 2018-07-13:
@sapan-ostic It looks like ipopt can't find blas.  Are you sure that your libipopt.so works as expected?  Did you follow the build stages at https://www.coin-or.org/Ipopt/documentation/node10.html?  Are you working on linux?  You could just apt install ipopt and it will handle the dependencies for you.  If you run into other problems, you could look at my fork https://github.com/jwdinius/ifopt.  I also wrote a blog post about the conversion process https://jwdinius.github.io/blog/2018/cmake/.  Hopefully one of these references will help.  Good luck!

guzhaoyuan, 2019-08-19:
I am having the same issue as @sapan-ostic.
I try to compile the [ex_test_ipopt.cc](https://github.com/ethz-adrl/ifopt/blob/master/ifopt_ipopt/test/ex_test_ipopt.cc) using cmake. 
```
find_package(IPOPT 3.11.10 REQUIRED)
find_package(ifopt 2.0.1 REQUIRED)
set(LIB_IPOPT ifopt_ipopt)
add_executable(${LIB_IPOPT}-example ex_test_ipopt.cc)
target_link_libraries(${LIB_IPOPT}-example
  PUBLIC 
    ifopt::ifopt_core
    ifopt::ifopt_ipopt
)
```
And it gives me:
```
[ 20%] Linking CXX executable ifopt_ipopt-example
//usr/local/lib/libipopt.so.1: undefined reference to `dpotrf_'
//usr/local/lib/libipopt.so.1: undefined reference to `dgemm_'
//usr/local/lib/libipopt.so.1: undefined reference to `dnrm2_'
//usr/local/lib/libipopt.so.1: undefined reference to `dscal_'
//usr/local/lib/libipopt.so.1: undefined reference to `dpotrs_'
//usr/local/lib/libipopt.so.1: undefined reference to `dtrsm_'
//usr/local/lib/libipopt.so.1: undefined reference to `dlopen'
//usr/local/lib/libipopt.so.1: undefined reference to `daxpy_'
//usr/local/lib/libipopt.so.1: undefined reference to `dgetrf_'
//usr/local/lib/libipopt.so.1: undefined reference to `dcopy_'
//usr/local/lib/libipopt.so.1: undefined reference to `idamax_'
//usr/local/lib/libipopt.so.1: undefined reference to `dlclose'
//usr/local/lib/libipopt.so.1: undefined reference to `dlerror'
//usr/local/lib/libipopt.so.1: undefined reference to `dlsym'
//usr/local/lib/libipopt.so.1: undefined reference to `dsyrk_'
//usr/local/lib/libipopt.so.1: undefined reference to `dasum_'
//usr/local/lib/libipopt.so.1: undefined reference to `dsyev_'
//usr/local/lib/libipopt.so.1: undefined reference to `dmumps_c'
//usr/local/lib/libipopt.so.1: undefined reference to `dgetrs_'
//usr/local/lib/libipopt.so.1: undefined reference to `ddot_'
//usr/local/lib/libipopt.so.1: undefined reference to `dsymv_'
//usr/local/lib/libipopt.so.1: undefined reference to `dgemv_'
collect2: error: ld returned 1 exit status
CMakeFiles/ifopt_ipopt-example.dir/build.make:96: recipe for target 'ifopt_ipopt-example' failed
make[2]: *** [ifopt_ipopt-example] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ifopt_ipopt-example.dir/all' failed
make[1]: *** [CMakeFiles/ifopt_ipopt-example.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

guzhaoyuan, 2019-08-19:
I solved this by reinstall the ipopt with BLAS. Aka, remove the `coinor-libipopt-dev`, and compile ipopt from source after get the Thirdparty solver BLAS.