ethz-adrl/ifopt #14: ifopt- ipopt problem

yousiri, 2018-06-11:
I have installed ipopt in /usr/local and it can be use directly. But when i use it in ifopt, some problems occur as the following (i have changed the part of "Find IPOPT" in the CMakeList.txt)
/usr/local/lib/libipopt.so: undefined reference to `dlopen'
/usr/local/lib/libipopt.so: undefined reference to `dlclose'
/usr/local/lib/libipopt.so: undefined reference to `dlerror'
/usr/local/lib/libipopt.so: undefined reference to `dlsym'

i want to get some help, thank you.

awinkler, 2018-06-11:
Possibly this independent of ifopt, but the library that is not found is "dl", necessary to dynamically load it. Please check the suggestions here: https://stackoverflow.com/questions/956640/linux-c-error-undefined-reference-to-dlopen

yousiri, 2018-06-12:
thank you for your response, i have see this link before, but i don't know where or how to load it?

awinkler, 2018-06-12:
You could try adding -ldl here:
https://github.com/ethz-adrl/ifopt/blob/f4174a8439fb8cfabdfe07a2a84d6e0bd8ed2b0b/ifopt_ipopt/CMakeLists.txt#L4

yousiri, 2018-06-12:
i have solve my problem, i put "CMAKE_DL_LIBS" into the "target_link_libraries" as the following: 

# The IPOPT solver interface
add_library(${PROJECT_NAME} src/ipopt_adapter.cc)
target_link_libraries(${PROJECT_NAME}
  ${IPOPT_LIBRARIES}
  ${ifopt_core_LIBRARIES}
  ${IPOPT_HSL_LIBRARIES}
  ${IPOPT_BLAS_LIBRARIES}
  ${IPOPT_LAPACK_LIBRARIES}
  ${IPOPT_MUMPS_LIBRARIES}
  ${IPOPT_METIS_LIBRARIES}
  ${CMAKE_DL_LIBS}
)

closed 2018-06-12 by yousiri