ethz-adrl/ifopt #23: Compiling ex_test_ipopt.cc gives error

x34903, 2018-07-12:
/tmp/ccexmy3Z.o: In function `main':
ex_test_ipopt.cc:(.text+0x136): undefined reference to `ifopt::Problem::Problem()'
ex_test_ipopt.cc:(.text+0x171): undefined reference to `ifopt::Problem::AddVariableSet(std::shared_ptr<ifopt::Component>)'
ex_test_ipopt.cc:(.text+0x1c1): undefined reference to `ifopt::Problem::AddConstraintSet(std::shared_ptr<ifopt::ConstraintSet>)'
ex_test_ipopt.cc:(.text+0x20e): undefined reference to `ifopt::Problem::AddCostSet(std::shared_ptr<ifopt::ConstraintSet>)'
ex_test_ipopt.cc:(.text+0x235): undefined reference to `ifopt::Problem::PrintCurrent() const'
ex_test_ipopt.cc:(.text+0x285): undefined reference to `ifopt::Ipopt::Solve(ifopt::Problem&)'
ex_test_ipopt.cc:(.text+0x29b): undefined reference to `ifopt::Problem::GetOptVariables() const'
/tmp/ccexmy3Z.o: In function `ifopt::Component::~Component()':
ex_test_ipopt.cc:(.text._ZN5ifopt9ComponentD2Ev[_ZN5ifopt9ComponentD5Ev]+0xd): undefined reference to `vtable for ifopt::Component'
/tmp/ccexmy3Z.o: In function `ifopt::ExVariables::ExVariables(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
ex_test_ipopt.cc:(.text._ZN5ifopt11ExVariablesC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN5ifopt11ExVariablesC5ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x21): undefined reference to `ifopt::VariableSet::VariableSet(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/tmp/ccexmy3Z.o: In function `ifopt::ExVariables::GetBounds() const':
ex_test_ipopt.cc:(.text._ZNK5ifopt11ExVariables9GetBoundsEv[_ZNK5ifopt11ExVariables9GetBoundsEv]+0x25): undefined reference to `ifopt::Component::GetRows() const'
/tmp/ccexmy3Z.o: In function `ifopt::ConstraintSet::~ConstraintSet()':
ex_test_ipopt.cc:(.text._ZN5ifopt13ConstraintSetD2Ev[_ZN5ifopt13ConstraintSetD5Ev]+0xd): undefined reference to `vtable for ifopt::ConstraintSet'
/tmp/ccexmy3Z.o: In function `ifopt::ExConstraint::ExConstraint(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
ex_test_ipopt.cc:(.text._ZN5ifopt12ExConstraintC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN5ifopt12ExConstraintC5ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x21): undefined reference to `ifopt::ConstraintSet::ConstraintSet(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/tmp/ccexmy3Z.o: In function `ifopt::ExConstraint::GetValues() const':
ex_test_ipopt.cc:(.text._ZNK5ifopt12ExConstraint9GetValuesEv[_ZNK5ifopt12ExConstraint9GetValuesEv]+0x25): undefined reference to `ifopt::Component::GetRows() const'
ex_test_ipopt.cc:(.text._ZNK5ifopt12ExConstraint9GetValuesEv[_ZNK5ifopt12ExConstraint9GetValuesEv]+0x97): undefined reference to `ifopt::Composite::GetComponent(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) const'
/tmp/ccexmy3Z.o: In function `ifopt::ExConstraint::GetBounds() const':
ex_test_ipopt.cc:(.text._ZNK5ifopt12ExConstraint9GetBoundsEv[_ZNK5ifopt12ExConstraint9GetBoundsEv]+0x25): undefined reference to `ifopt::Component::GetRows() const'
/tmp/ccexmy3Z.o: In function `ifopt::ExConstraint::FillJacobianBlock(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, Eigen::SparseMatrix<double, 1, int>&) const':
ex_test_ipopt.cc:(.text._ZNK5ifopt12ExConstraint17FillJacobianBlockENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERN5Eigen12SparseMatrixIdLi1EiEE[_ZNK5ifopt12ExConstraint17FillJacobianBlockENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERN5Eigen12SparseMatrixIdLi1EiEE]+0x92): undefined reference to `ifopt::Composite::GetComponent(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) const'
/tmp/ccexmy3Z.o: In function `ifopt::CostTerm::~CostTerm()':
ex_test_ipopt.cc:(.text._ZN5ifopt8CostTermD2Ev[_ZN5ifopt8CostTermD5Ev]+0xd): undefined reference to `vtable for ifopt::CostTerm'
/tmp/ccexmy3Z.o: In function `ifopt::ExCost::ExCost(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
ex_test_ipopt.cc:(.text._ZN5ifopt6ExCostC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN5ifopt6ExCostC5ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x1f): undefined reference to `ifopt::CostTerm::CostTerm(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/tmp/ccexmy3Z.o: In function `ifopt::ExCost::GetCost() const':
ex_test_ipopt.cc:(.text._ZNK5ifopt6ExCost7GetCostEv[_ZNK5ifopt6ExCost7GetCostEv]+0x68): undefined reference to `ifopt::Composite::GetComponent(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) const'
/tmp/ccexmy3Z.o: In function `ifopt::ExCost::FillJacobianBlock(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, Eigen::SparseMatrix<double, 1, int>&) const':
ex_test_ipopt.cc:(.text._ZNK5ifopt6ExCost17FillJacobianBlockENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERN5Eigen12SparseMatrixIdLi1EiEE[_ZNK5ifopt6ExCost17FillJacobianBlockENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERN5Eigen12SparseMatrixIdLi1EiEE]+0x92): undefined reference to `ifopt::Composite::GetComponent(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) const'
/tmp/ccexmy3Z.o: In function `ifopt::Ipopt::Ipopt()':
ex_test_ipopt.cc:(.text._ZN5ifopt5IpoptC2Ev[_ZN5ifopt5IpoptC5Ev]+0x1a): undefined reference to `vtable for ifopt::Ipopt'
/tmp/ccexmy3Z.o:(.rodata._ZTVN5ifopt6ExCostE[_ZTVN5ifopt6ExCostE]+0x20): undefined reference to `ifopt::CostTerm::GetValues() const'
/tmp/ccexmy3Z.o:(.rodata._ZTVN5ifopt6ExCostE[_ZTVN5ifopt6ExCostE]+0x28): undefined reference to `ifopt::CostTerm::GetBounds() const'
/tmp/ccexmy3Z.o:(.rodata._ZTVN5ifopt6ExCostE[_ZTVN5ifopt6ExCostE]+0x38): undefined reference to `ifopt::ConstraintSet::GetJacobian() const'
/tmp/ccexmy3Z.o:(.rodata._ZTVN5ifopt6ExCostE[_ZTVN5ifopt6ExCostE]+0x40): undefined reference to `ifopt::Component::Print(double) const'
/tmp/ccexmy3Z.o:(.rodata._ZTVN5ifopt12ExConstraintE[_ZTVN5ifopt12ExConstraintE]+0x38): undefined reference to `ifopt::ConstraintSet::GetJacobian() const'
/tmp/ccexmy3Z.o:(.rodata._ZTVN5ifopt12ExConstraintE[_ZTVN5ifopt12ExConstraintE]+0x40): undefined reference to `ifopt::Component::Print(double) const'
/tmp/ccexmy3Z.o:(.rodata._ZTVN5ifopt11ExVariablesE[_ZTVN5ifopt11ExVariablesE]+0x40): undefined reference to `ifopt::Component::Print(double) const'
/tmp/ccexmy3Z.o:(.rodata._ZTVN5ifopt11VariableSetE[_ZTVN5ifopt11VariableSetE]+0x40): undefined reference to `ifopt::Component::Print(double) const'
/tmp/ccexmy3Z.o:(.rodata._ZTIN5ifopt6ExCostE[_ZTIN5ifopt6ExCostE]+0x10): undefined reference to `typeinfo for ifopt::CostTerm'
/tmp/ccexmy3Z.o:(.rodata._ZTIN5ifopt12ExConstraintE[_ZTIN5ifopt12ExConstraintE]+0x10): undefined reference to `typeinfo for ifopt::ConstraintSet'
/tmp/ccexmy3Z.o:(.rodata._ZTIN5ifopt11VariableSetE[_ZTIN5ifopt11VariableSetE]+0x10): undefined reference to `typeinfo for ifopt::Component'
/tmp/ccexmy3Z.o: In function `ifopt::Composite::~Composite()':
ex_test_ipopt.cc:(.text._ZN5ifopt9CompositeD2Ev[_ZN5ifopt9CompositeD5Ev]+0xd): undefined reference to `vtable for ifopt::Composite'
/tmp/ccexmy3Z.o: In function `ifopt::Ipopt::~Ipopt()':
ex_test_ipopt.cc:(.text._ZN5ifopt5IpoptD2Ev[_ZN5ifopt5IpoptD5Ev]+0xd): undefined reference to `vtable for ifopt::Ipopt'
collect2: error: ld returned 1 exit status

awinkler, 2018-07-13:
Hi! Are you building ifopt with standard cmake or catkin according to the README.md instructions?

sapan-ostic, 2018-07-13:
Hi @awinkler ,
I am following this issue on-behalf of @x34903. We are working on the same task. 
I am building ifopt with catkin.  adding the ifopt package in the CMakeLists.txt solve the problem.
Thanks.

awinkler, 2018-07-14:
Great, glad this worked, so closing this issue for now. Please re-open if something is still unclear.

closed 2018-07-14 by awinkler