ethz-adrl/ifopt #24: Test fails for building with Catkin

sapan-ostic, 2018-07-13:
**~/catkin_ws/build$ ./ifopt/ifopt_core/ifopt_core-test**
bash: ./ifopt/ifopt_ipopt/ifopt_ipopt-example: No such file or directory

awinkler, 2018-07-13:
Hi @sapan-ostic,
Did you run these commands from your `catkin_ws/build` folder? Check that folder and search for the executable `ifopt_core-test`, possibly it's in a different location.

sapan-ostic, 2018-07-13:
Hi,
Oh yes, I found the ifopt_ipopt-example executable file in catkin_ws/devel_isolated/ifopt/bin
But I am getting Segmentation fault (core dumped) on executing it.
I am trying to run ifopt_ipopt-example.

awinkler, 2018-07-14:
It's possible that your IPOPT version installed through your package manager is smaller than 3.11.4 (default in UBUNTU 14.04). This has a bug related to mumps and will segfault. If this is the case you should try installing a different IPOPT version as described here:
http://docs.ros.org/kinetic/api/ifopt/html/#install

closed 2018-10-07 by awinkler