ethz-adrl/ifopt #29: Example on vector CostTerm ?

artivis, 2018-07-27:
Hi,
thanks for this interesting project !

I was wondering if this project allows to write `CostTerm`s which cost is a vector ??  
e.g. ` e = var_set1 - var_set0 ` a vector element-wise subtraction (`e, var_set0, var_set1` in R^n).

[This picture](https://github.com/ethz-adrl/ifopt/blob/master/doc/ifopt.png) in the doc seems to indicate that it is doable but with `CostTerm::GetCost` return type being a `double` I can't get my head around this issue.

Could you please provide some more info, possibly an example ??

Cheers.

awinkler, 2018-07-27:
Hi!
In the end, the cost always has to be scalar, that's just the way nonlinear programming problems are defined. This cost however can be composed of a variety of scalar cost-terms. You can subtract a scalar cost term by defining the costs as negative. 

The image shows the scalar cost J(x), and the gradient of this cost w.r.t x, which are two added 1xn row vectors (n being the number of optimization variables). Does that clear things up?

artivis, 2018-07-27:
Thanks for the (very) quick answer !

From a theoretical point of view we're on the same page. My question however was more from a programming point of view.  
Using the same example as before, say I would like to do this vector subtraction (or whatever the operation) all at once. To benefit from vectorization, because the operation is heavy, well whatever the reason... How could one somewhat 'pre-compute' the error (in its vector form) to later on 'distribute' each scalar cost-term ? Or how to 'group' this variety of `CostTerm` objects so that the operation is computed only once ?  
I'm not sure this comment make things clearer, hopefully.

artivis, 2018-07-27:
Well this does not fully answer my question but I shall dig in the code a little more and come back to you with a clearer idea.  
Thanks a lot anyway !

closed 2018-10-07 by awinkler