ethz-adrl/ifopt #34: Problems that contain no constraints segfault

wxmerkt, 2019-01-15:
Hi Alexander,
Thank you very much for providing this great wrapper! I am in the process of integrating it into our motion planning framework (in order to use Ipopt) and came across the following segmentation fault:

- If an NLP contains only a cost term, no constraint terms: it segfaults.
- If a 0-dimensional constraint is added: no segfault.

Best wishes,
Wolfgang

awinkler, 2019-01-16:
Thanks for pointing this out :+1: . Somewhere the code is probably trying to access the first element of an empty constraint-set vector. I'll look into it asap. If you have time, you could also you figure out where and propose a PR fix :)

wxmerkt, 2019-01-16:
The somewhere is the `GetJacobian()` call (`components_.first()`)

closed 2019-01-16 by awinkler