ethz-adrl/ifopt #38: Return codes for Solver

disRecord, 2019-02-08:
[`Solver` interface](https://github.com/ethz-adrl/ifopt/blob/8cc91688935867a67daeb39a8ff9ddb3f9750668/ifopt_core/include/ifopt/solver.h#L57)  does not provides any means to check solver exit code. 

I can see two ways to solve this issue:
1. Add solver specific methods to `IpoptSolver` and `SnoptSolver` classes. (it can be something like [this](https://github.com/sweetie-bot-project/ifopt/blob/7e3b82bbf08e86d077c5400f8a2344f984260d62/ifopt_ipopt/include/ifopt/ipopt_solver.h#L73)
2. Introduce a set of "common exit codes". So solver specific exit code is translated to common  exit code which can be returned from `Solve()`.

awinkler, 2019-02-09:
You're right, so far the solver exit codes are only printed. But we can also return these in the Solve(...) method, instead of void.
https://github.com/ethz-adrl/ifopt/blob/8cc91688935867a67daeb39a8ff9ddb3f9750668/ifopt_core/include/ifopt/solver.h#L57

So ipopt_solver and snopt_solver need to additionally return the solver specific status
I don't think we need the extra level of indirection with the common exit codes, let's start with this.

disRecord, 2019-02-09:
Yes, I think it's Ok. I tried to make a list of common exit statuses and it does not look good. WI

awinkler, 2019-02-10:
Can you please propose a PR? That would be great.

awinkler, 2020-10-05:
This can now be achieved using:

https://github.com/ethz-adrl/ifopt/blob/8d0e0a8a3c000c5964e7e6d422310dde1f249c3b/ifopt_core/include/ifopt/solver.h#L64

closed 2020-10-05 by awinkler