ethz-adrl/ifopt #39: Generic version of IPOPT

jbensabat, 2019-02-11:
Hello 
IPOPT comes with a number of interfaces to linear solvers (MUMPS, LAPACK, HSL) and others. 
I tried to create a dll of ipopt with none of these solvers in order to focus o Eigen. 
the build fails because it looks for these libs. 
Is there a clean version of IPOPT ? i.e., one that does not depend on any solver a priori ?
then the build of IPOPT+EIGEN could be much cleaner and simpler.
best
jac

awinkler, 2019-02-12:
The one you install from the Linux command line (sudo apt install coinor-libipopt-dev) has Mumps already linked and works out of the box. There's nothing else do be done or configured.

jbensabat, 2019-02-13:
Thanks 
the problem is that I work under windows and MSVC
I have found a CMAKE build of IPOPT

jbensabat, 2020-04-16:
hi 
I am also interested in building IPOPT on MSVS 
could you please indicate where to find the CMAKE build ?
thanks 
jac

awinkler, 2020-10-05:
I haven't tried building on Windows. But you should be able to generate the vsproject files (.sln, vcxproj) by running `cmake` on windows. Then, if you installed the additional c++ build tools with visual studio, you could use `msbuild` from the command line on the vsproject files. (This link might be helpful: https://dmerej.info/blog/post/cmake-visual-studio-and-the-command-line/)

But as you correctly pointed out, the problem will be for cmake to find and link against the installed version of IPOPT. And you will need at least one solver (e.g. MUMPS) to do anything, that's why it's so convenient that `sudo apt install coinor-libipopt-dev` installs that automatically on ubuntu.

Unfortunately there is no plan on adding windows support at this time.

closed 2020-10-05 by awinkler