ethz-adrl/ifopt #41: Adding Hessians to Component

mpowelson, 2019-05-20:
I notice that components only include values, bounds, and jacobians. Is there any reason Hessians are not stored in them as well, other than you just didn't need it?

We are considering using this library for a project, and a lot of the solvers we'd like to interface to accept hessians. I'm not entirely sure how this would look. I suppose it could actually go in CostTerm?

awinkler, 2019-05-20:
Yes, I just didn't use them, mostly because it was quite difficult to determine their analytical values for the problem I was solving. But they could be included in ifopt as well. 

From the structure I would add a function "GetHessian()", that's callable on a cost-term, as well as on a constraint-set. That would be the 2nd derivative (3 dimensional matrix, right?), just as the function "GetJacobian()" supplies the first derivative (2d). The image here at the bottom might help: http://docs.ros.org/api/ifopt/html/group__ProblemFormulation.html

mpowelson, 2019-05-20:
Ok that makes sense. I was not sure about the 3D matrix. I suppose for constraints it would have to be. I'm not an optimization guy. It seems like many of them only really care about the hessian of the objective function which could then just be 2D. But I could be completely wrong about that.

jmainpri, 2019-12-16:
Hi, any progress here? 

My understanding of non-linear interior-point optimization is that you only reason on 2d order derivatives for the objective, which is a simple matrix. The jacobian of the objective would actually be just the transpose of the gradient, so a vector. The same should apply for SQP:  "SQP methods solve a sequence of optimization subproblems, each of which optimizes a quadratic model of the objective subject to a linearization of the constraints" (cf. wikipedia). 

To have exact Hessians passed in, what would one need to change to ifopt?

 Thanks in advance.

mpowelson, 2019-12-16:
While I am still interested, I have not looked into implementing this any more.

graphitical, 2020-10-19:
I went looking for this capability today since the problem I'm working on can efficiently compute an analytical Hessian matrix and I'm enjoying the ease of use ifopt provides. It seems like the current status is still open so I started to take a stab at it, naively searching for the word "Jacobian" and extending the code as needed. It will take some further investigation to make sure there aren't any nuances I skipped over.

[This link](https://coin-or.github.io/Ipopt/INTERFACES.html) indicates that the Hessian of the whole Lagrangian is need, that is the Hessian of the objective function _and_ the constraints so both Components will need to be extended. From what I understand they should all be simple matrices (rank-2 tensors) as well.

IoannisDadiotis, 2021-06-08:
Hi all,

Any update on this? Did somebody implement this and wants to provide the code extensions needed?