ethz-adrl/ifopt #42: IPOPT's maximum cpu time is not reset when violated

orsoromeo, 2019-07-02:
Hi everyone,
I noticed that, whenever IPOPT is not able to find a solution because the MAXIMUM_CPU_TIME is exceeded, this termination mode will keep on being triggered forever unless you restart IPOPT. This means that if, for example, I attempt to optimize a solution and the solver does not find a solution because of the maximum_cpu_time, then the solver won't find a solution even for the next formulations, even if I change robot, terrain or any other possible configuration. It really looks like the MAXIMUM_CPU_TIME flag of IPOPT is not properly reset and, when triggered, remains always ON.
Please notice that this is probably an IPOPT bug, but I am wondering if anyone here noticed the same issue and found a fix for it.

awinkler, 2019-07-08:
Hi! Thanks for pointing this out. I have not experienced this issue, but let me know if you figure anything out!

orsoromeo, 2019-07-30:
Hi Alex, I am surprised you never encountered this issue, it seems to always show up whenever you want to do some replanning with TOWR. One workaround is of course to set the maximum cpu time to a very high value but that goes against the idea of replanning :)