ethz-adrl/ifopt #44: Provide only part of derivatives to SNOPT

mzwang828, 2020-02-24:
First of all, thanks a lot for providing us this super convenient tool!
I'm using it for constructing problems for Snopt. From the document of Snopt, I read that the user can provide part of gradients and Snopt can estimate the left gradients by finite differences. When I'm using ifopt, it seems to me that the Jacobian is first initialized as all zeros and then certain values are substituted by calculated derivates. Snopt thus considers that all the gradients are provided and will not estimate anything. I'm wondering is there's a way that I can calculate part of the derivates in my code and inform Snopt that the left derivates need estimation.

mzwang828, 2020-02-26:
I have figured out how to do this in a "manual" way.
To make SNOPT estimate the not provided derivatives, I first need to calculate the number of missing terms and their positions in the final Jacobian. I then have to modify the neG, iGfun and iGvar in snopt_adapter.cc accordingly to make SNOPT aware of missing terms.
Thanks again for this convenient tool!

closed 2020-02-26 by mzwang828

awinkler, 2020-02-27:
Thanks @mzwang828 for looking into this and posting your solution!