ethz-adrl/ifopt #45: install process, make test error

NUDTUGVexplorer, 2020-03-05:
Hi, I found an error before install.
After make, when I run make test, the issues are shown as follows:
Running tests...
Test project /home/lxh/software/ifopt-master/build
    Start 1: ifopt_core-test
1/2 Test #1: ifopt_core-test ..................   Passed    0.00 sec
    Start 2: ifopt_ipopt-example
2/2 Test #2: ifopt_ipopt-example ..............***Exception: Other  0.09 sec

50% tests passed, 1 tests failed out of 2

Total Test time (real) =   0.09 sec

The following tests FAILED:
	  2 - ifopt_ipopt-example (OTHER_FAULT)
Errors while running CTest
make: *** [test] error 8

My OS:
OS: ubuntu 14.04 LTS, and gcc g++ version are 4.8.4

awinkler, 2020-03-18:
Hi! I know there used to be an issue installing IPOPT directly through ubuntu 14.04. (sudo apt-get install coinor-libipopt-dev). You could try installing/building ipopt from source, if that's where the error is thrown from. In 16.04 this should work out of the box.

NUDTUGVexplorer, 2020-03-18:
Thanks Awinkler. I fixed that by reinstalling the Ipopt library. The third-party library of Ipopt, Mumps is not correctly installed. After I download the source code of the Mumps, and recompile the ipopt library. The problem is fixed. Anyway, thanks for your advise to upgrade the Ubuntu to be 16.04. I guess it is another way to fix the problem.

awinkler, 2020-03-18:
Great, glad this helped!

closed 2020-03-18 by awinkler