ethz-adrl/ifopt #46: IPOPT finite difference result is different from exact

paulyang1990, 2020-03-10:
I did a test using the example code provided in ipopt test to compare the result of using "ipopt.SetOption("jacobian_approximation", "exact"); " and using "ipopt.SetOption("jacobian_approximation", "finite-difference-values");"

If I use the exact jacobian provided in the code, I got optimal solution be "1.00 0.00". If I change the jacobian option to finite difference. Nothing changes. I am wondering the algorithm is still using the provided jacobian. So I comment out the content of functions "FillJacobianBlock" of ExCost and ExConstraint. Now the optimal solution I got is "0.44 0.81", which has lower cost but slightly violate the constraint. If I change the initial condition to be "1.5 0.5", then the optimal solution is "0.81 0.34", still violates the constraint. I guess the finite difference method suffers from numerical problem in this case. 

Could anyone help me to answer following questions:
1. Whether I am using "finite-difference-values" correctly or not. 
2. How to improve the numerical precision of finite difference method.

mzwang828, 2020-09-07:
Hi @paulyang1990 , I got exactly the same behavior when I'm using "finite-difference-values" option. Do you finally manage to solve this?
Since the gradients in my problem are quite difficult to write, I'd really appreciate any insights on this issue.

pcdangio, 2020-10-04:
I was doing some testing, and I noticed that "finite-difference-values" only applies to constraint_sets, and not cost_terms.

I put cout statements in the FillJacobianBlock methods for the ExConstraint and ExCost classes.  With "finite-difference-values" enabled, FillJacobianBlock is NOT called for ExConstraint, but IS called for ExCost.

If you commented out the ExCost::FillJacobianBlock implementation, it breaks the problem because "finite-difference-values" does not apply to the cost term jacobian calculation.  Hence why the incorrect answer of "0.44 0.81" is returned.

I can't find a way to force Ipopt to use finite difference values on the cost term's jacobian calculation.

awinkler, 2020-10-05:
Thanks for pointing this bug out. Setting `ipopt.SetOption("jacobian_approximation", "finite-difference-values")`  now gives the correct results without needing to define and derivatives/jacobians  (see #61). More precisely: there is no need to `FillJacobianBlock()` of the constraint _nor_ the cost term when using this option in IPOPT. 

For future reference, this option only works when choosing the IPOPT solver. It is not yet implemented for SNOPT.

closed 2020-10-05 by awinkler

pcdangio, 2020-10-05:
Awesome, thanks for the speedy fix!  Follow up question - how long do your patches usually take to trickle into the ros-<distro>-ifopt binaries?

awinkler, 2020-10-05:
Sure thing. I have to see if it makes sense to tag this version and create a new release. That's independent of pushing to the master branch.  Let's give it a few days to see if any issues come up and fix them first.

prabinrath, 2021-06-19:
> Awesome, thanks for the speedy fix! Follow up question - how long do your patches usually take to trickle into the ros--ifopt binaries?

I am facing the same problem with ros binaries. Is it not yet released?

PR #61 (2020-10-05): calculate finite difference of cost-term for IPOPT if flag set.
resolves #46