ethz-adrl/ifopt #47: Runtime error: Assertion `nele == nele_jac' failed

RobertoRoos, 2020-04-08:
Ran into this problem while doing some optimisations on dynamic systems with IFOPT (not Towr, but something very similar).

When the optimisation starts I get the message:

> Gambol: */fopt/ifopt_ipopt/src/ipopt_adapter.cc:139: virtual bool Ipopt::IpoptAdapter::eval_jac_g(Ipopt::Index, const double*, bool, Ipopt::Index, Ipopt::Index, Ipopt::Index*, Ipopt::Index*, double*): Assertion `nele == nele_jac' failed.`

This for a problem with an exact constraint jacobian.

I looked at the source and saw this assertion fails when the number of assigned elements in the jacobian construction does not equal the expected number of non-zeros (`nele_jac`).  
I also noticed that IFOPT finds `nele_jac` inside `get_nlp_info()` by executing the constraint jacobian and looking at the result. This makes it all the more strange that I should encounter this problem.

The jacobian section is case is being built by multiplying a fixed SparseMatrix with a jacobian calculated by a variable itself. (Vague, I know, but I'm a bit too embarrassed to share the entire code.)

Thank you!

RobertoRoos, 2020-04-08:
Found the cause. When calculating the jacobian Eigen::Dense matrix is calculated first. This is converted with `.sparseView()` to a Eigen::Sparse matrix such that it can be multiplied for the jacobian section. However, in the initial guess for the optimisation there just happened to be zeros in the dense matrix, which were skipped in the sparse conversion. _TL;DR_, the number of non-zeroes in the jacobian was not constant.

I avoided this by not using `sparseView()`, but instead doing a manual conversion such that zero values were still included and the number of non-zeroes remains constant.

@awinkler I'll leave it up to you close / delete this. Thanks!

evouga, 2020-08-24:
I just ran into the same bug. It's insidious because the initial guess is often going to have more "accidental zeros" than a later configuration during optimization.

It would be nice to put a big warning in the documentation that FillJacobianBlock *must* always return a sparse matrix with the same sparsity pattern (even if some entries happen to be zero).

awinkler, 2020-08-24:
thanks for pointing this out. I hope the added comments make this clearer now.

closed 2020-08-24 by awinkler