ethz-adrl/ifopt #60: Redundant calls to FillJacobianBlock

pcdangio, 2020-10-04:
During debugging, I noticed that FillJacobianBlock is called on each constraint/cost twice, with the second call being redundant.

I placed cout statements in each constraint/cost FillJacobianBlock statements, and ran using an IpoptSolver. The output shows:

```
FillJacobianBlock on cost_term_1 for var_set var_set1
0.82477262908425008625
0.34705019768468647889
FillJacobianBlock on cost_term_1 for var_set var_set1
0.82477262908425008625
0.34705019768468647889
FillJacobianBlock on constraint_term_1 for var_set var_set1
0.82477262908425008625
0.34705019768468647889
FillJacobianBlock on constraint_term_1 for var_set var_set1
0.82477262908425008625
0.34705019768468647889
FillJacobianBlock on cost_term_1 for var_set var_set1
0.91151018263754624016
0.20160334655763342826
FillJacobianBlock on cost_term_1 for var_set var_set1
0.91151018263754624016
0.20160334655763342826
FillJacobianBlock on constraint_term_1 for var_set var_set1
0.91151018263754624016
0.20160334655763342826
FillJacobianBlock on constraint_term_1 for var_set var_set1
0.91151018263754624016
0.20160334655763342826
FillJacobianBlock on cost_term_1 for var_set var_set1
 0.99911511172637545997
0.031011668948078191521
FillJacobianBlock on cost_term_1 for var_set var_set1
 0.99911511172637545997
0.031011668948078191521
FillJacobianBlock on constraint_term_1 for var_set var_set1
 0.99911511172637545997
0.031011668948078191521
FillJacobianBlock on constraint_term_1 for var_set var_set1
 0.99911511172637545997
0.031011668948078191521
FillJacobianBlock on cost_term_1 for var_set var_set1
   0.99999116101726370776
0.00012925427351325596836
FillJacobianBlock on cost_term_1 for var_set var_set1
   0.99999116101726370776
0.00012925427351325596836
FillJacobianBlock on constraint_term_1 for var_set var_set1
   0.99999116101726370776
0.00012925427351325596836
FillJacobianBlock on constraint_term_1 for var_set var_set1
   0.99999116101726370776
0.00012925427351325596836
FillJacobianBlock on constraint_term_1 for var_set var_set1
   0.99999718292395767261
5.6341804121589472969e-06
FillJacobianBlock on constraint_term_1 for var_set var_set1
   0.99999718292395767261
5.6341804121589472969e-06
FillJacobianBlock on cost_term_1 for var_set var_set1
   0.99999718292395767261
5.6341804121589472969e-06
FillJacobianBlock on cost_term_1 for var_set var_set1
   0.99999718292395767261
5.6341804121589472969e-06
```

awinkler, 2020-10-05:
thanks for pointing this out. Could you post the code you used to generate the print-outs and I'll try and recreate it and have a look.  Thanks :)

pcdangio, 2020-10-07:
This is using the example code from:

`ifopt/ifopt_ipopt/test/ex_test_ipopt.cc`

with the problem defined in:

`ifopt/ifopt_core/test/ifopt/test_vars_constr_cost.h`

The only thing I changed was adding the a cout statement in the FillJacobianBlock methods for both the constraint_set and cost_term:

**constraint_set "constraint1":**
```
  void FillJacobianBlock (std::string var_set, Jacobian& jac_block) const override
  {
    // must fill only that submatrix of the overall Jacobian that relates
    // to this constraint and "var_set1". even if more constraints or variables
    // classes are added, this submatrix will always start at row 0 and column 0,
    // thereby being independent from the overall problem.
    if (var_set == "var_set1") {
      Vector2d x = GetVariables()->GetComponent("var_set1")->GetValues();

      // COUT STATEMENT FOR DEBUGGING
      std::cout << "FillJacobianBlock on constraint_set_1 for var_set " << var_set << std::endl << x << std::endl;

      jac_block.coeffRef(0, 0) = 2.0*x(0); // derivative of first constraint w.r.t x0
      jac_block.coeffRef(0, 1) = 1.0;      // derivative of first constraint w.r.t x1
    }
  }
```

**cost_term "cost_term1":**
```
  void FillJacobianBlock (std::string var_set, Jacobian& jac) const override
  {
    if (var_set == "var_set1") {
      Vector2d x = GetVariables()->GetComponent("var_set1")->GetValues();

      // COUT STATEMENT FOR DEBUGGING
      std::cout << "FillJacobianBlock on cost_term_1 for var_set " << var_set << std::endl << x << std::endl;

      jac.coeffRef(0, 0) = 0.0;             // derivative of cost w.r.t x0
      jac.coeffRef(0, 1) = -2.0*(x(1)-2.0); // derivative of cost w.r.t x1
    }
  }
```

Levi-Armstrong, 2021-01-11:
Just curious, was a conclusion made?

awinkler, 2021-01-18:
This was not ideal and is now improved: see https://github.com/ethz-adrl/ifopt/commit/a44204fd1477906f1a7d0cb90899036869c9c60e

So during every optimization iteration, the jacobian of *only the first* contraint_set and first cost-term was called twice (the first time to determine the number of variables, the second time to evaluate it). This is improved now, by making the variable `static`, so this n_var call is only made once during the entire optimization (not every iteration). This will make things faster if you had only a small number of constraint_sets or your first constraint_set was very costly to evaluate.

Thanks for discovering this @pcdangio and @Levi-Armstrong for following up. 👍

JohannesMannheim, 2021-01-21:
Thank you all for the effort, I have experienced this issue as well.
 Sadly, the new version does not work for me. I get a segfault in 
`Ipopt::IpoptAdapter::eval_grad_f(int, double const*, bool, double*) () from libifopt_ipopt.so`

Manually removing `static` in `composite.cc` prevents the segfault.
Does this problem arise only in my specific setup?

awinkler, 2021-01-21:
You're right, this doesn't seem to work when the constraints are empty. I'll look into fixing this. Thanks for pointing this out.

Levi-Armstrong, 2021-01-21:
We are also experiencing the same behavior. Thank you for looking into it.

awinkler, 2021-01-21:
Fixed in https://github.com/ethz-adrl/ifopt/commit/27451ce3212d3c0ccd3a0db9a1ee9a9963466674.
Please let me know if you see other issues.

Levi-Armstrong, 2021-01-21:
It resolved the issues we were having. Thank you for the fix.

JohannesMannheim, 2021-01-22:
Same here.  It works now. Only in the first iteration, the Jacobian is evaluated twice. Thank you.

Levi-Armstrong, 2022-01-22:
This looks to now be resolved. Reopen if need.

closed 2022-01-22 by Levi-Armstrong