ethz-adrl/ifopt #66: ROS 2 Foxy Release?

Levi-Armstrong, 2021-01-11:
I noticed that this package has releases on most of the ROS 1 distros, but did not see anything on the ROS 2 distros. Are there plans to release this on ROS 2 Foxy?

awinkler, 2021-01-18:
Wasn't on my radar, but will think about it if people would find this useful. Will ROS1 be deprecated at some point?

Levi-Armstrong, 2021-01-18:
I believe Noetic is the last release of ROS 1. We currently use IFOPT on ROS 2 just building from source and the release process is very similar.

vatanaksoytezer, 2021-06-17:
@awinkler we would also appreciate a Foxy release, this would mean one less source build for all repos using ifopt.

DLu, 2021-11-03:
@awinkler +1 for Foxy release. Happy to help as needed

awinkler, 2021-11-03:
Sorry for being unresponsive, I'm a bit swamped right now. But happy to give you write access to the relevant repos to create the required ROS releases, if that's something you'd be willing to do.

DLu, 2021-11-09:
https://github.com/ros/rosdistro/pull/31177

awinkler, 2021-11-10:
awesome, thanks @DLu!

closed 2021-11-10 by DLu

Levi-Armstrong, 2021-11-12:
@DLu Did you also do an updated noetic release?

DLu, 2021-11-16:
Sorry, I was too slow to get to this. Thanks for releasing it @Levi-Armstrong