ethz-adrl/ifopt #67: Question about warm starting

junhyeokahn, 2021-01-17:
Hello!
I appreciate your efforts in maintaining this nice repo.

I have a quick question about the warm starting of the optimizer.
Can I just set variables with the numbers I want to initialize?
Or should I call additional method such as ```SetOption("warm_start_init_point", "yes"); ```?

awinkler, 2021-01-18:
Hi! It works without the option, although IPOPT might do some smarter tricks when you additionally set that option, I haven't experimented with that yet.

Generally I've found it's quite difficult to warm-start interior point methods like IPOPT, because of the way they solve the problem: The initial variables are often changed a lot to move into the center of the constraints during the first iteration. SQP methods, like SNOPT, might be a better choice if you have trustworthy information of the initial values (like in Model-Predictive-Control, where the initial values can be the solution of the previous solve).

junhyeokahn, 2021-01-18:
Thanks for your kind reply!

closed 2021-01-18 by junhyeokahn

mzwang828, 2021-02-11:
> Hi! It works without the option, although IPOPT might do some smarter tricks when you additionally set that option, I haven't experimented with that yet.
>
> Generally I've found it's quite difficult [...]

Hi @awinkler, I have the same question about using SNOPT. Do I just set the variables with good initial values or I should also change the parameter "Cold" in the snopt_sovler.cc? Once I set Cold = 2, which stands for warm starting, SNOPT gives warnings of "Invalid argument Fstate" and it seems that Fstate are not assigned with any meaningful values in SnoptAdapter. Thank you!