ethz-adrl/ifopt #70: The EvaluateCostFunctionGradient numerical difference produces incorrect gradient when using squared cost.

Levi-Armstrong, 2021-05-21:
I believe this is the reason IPOPT using 1e-8 due to floating-point precision when having squared cost. If you paste the code into the [godbolt](https://godbolt.org/) compiler you get the output below which causes the gradient to be zero using the current epsilon. I plan to submit a PR allowing the user to set the epsilon.

``` c++
int main(int argc, char** argv)
{
    std::cout.precision(17);
    {
        double x = 3 + 1e-16;
        double x2 = x * x;
        std::cout << std::fixed << x2 << std::endl;
    }

    {
        double x = 3 + 1e-8;
        double x2 = x * x;
        std::cout << std::fixed << x2 << std::endl;
    }
}
```

output:

``` c++
ASM generation compiler returned: 0
Execution build compiler returned: 0
Program returned: 0
9.00000000000000000
9.00000005999999964
```

facontidavide, 2021-11-23:
Yes, I precisely had this problem today (it took me a while to debug it).

 I think this issue should be solved ASAP, it really makes the **finite-difference** option unusable.

Levi-Armstrong, 2021-11-29:
It has been updated so you may provide your own epsilon for the function but left the default value as per the authors request.

Levi-Armstrong, 2021-11-29:
Addressed by #71

closed 2021-11-29 by Levi-Armstrong