ethz-adrl/ifopt #75: errors with Snopt 7.7

basitimran37, 2021-12-03:
```make test``` passes for both snopt and ipopt but when I use ```SnoptSolver solver``` and ```solver.Solve(nlp)``` in my code instead of ```IpoptSolver```, it throws the following errors:

undefined reference to 'vtable for ifopt::SnoptSolver'
undefined reference to 'ifopt::SnoptSolver::Solve(ifopt::Problem&)

I have already followed the suggestions in #31 to make SNOPT work.

basitimran37, 2021-12-03:
This was just a missing reference to snopt in CmakeLists.txt of my project. Resolved!

closed 2021-12-03 by basitimran37

shbang91, 2022-05-14:
Hello, 
I am using Snopt 7.7 and build the ifopt_snopt successfully, but still `make test` failed. Could you share what modification you made?

I've already followed [https://github.com/ethz-adrl/ifopt/issues/31](url).

Thank you.