ethz-adrl/ifopt #82: Internal representation of optimization variables

naguil, 2022-05-12:
The idea of an internal representation sounds very useful. And I pretend to use something as polynomials or a Fourier series. However, I can't see how to access this representation from the Cost class, for example. For that reason, I fail to see its overall usefulness.

Could anyone exemplify the use of the internal representation of the optimization variables?

Levi-Armstrong, 2022-05-12:
When you create a problem assign variable sets like what is shown below with unique names. Then when you add a constraint or cost to the problem it get linked to the problems variable set. So inside the cost or constraint object you have access to the member function  `GetVariables()` which gives you access to all variable sets. Then during the solve it will call the `GetValues()` and `FillJacobianBlock`. I also included an example of a constraint implementation which may help.

``` c++
  Problem nlp;
  nlp.AddVariableSet(std::make_shared<ExVariables>("var_set0"));
  nlp.AddVariableSet(std::make_shared<ExVariables>("var_set1"));
```

Example:
https://github.com/tesseract-robotics/trajopt/blob/master/trajopt_ifopt/include/trajopt_ifopt/constraints/joint_velocity_constraint.h
https://github.com/tesseract-robotics/trajopt/blob/master/trajopt_ifopt/src/constraints/joint_velocity_constraint.cpp

naguil, 2022-05-12:
Thank you Levi. But, as I understand, you are getting the values of the external representation (as a VectorXd). I was hoping I could get the internal representation mentioned in the following code snippet. I'd like to define some x0_ and x1_ as Fourier series, for example.

`  // Here is where you can transform the Eigen::Vector into whatever
  // internal representation of your variables you have (here two doubles, but
  // can also be complex classes such as splines, etc..
  void SetVariables(const VectorXd& x) override
  {
    x0_ = x(0);
    x1_ = x(1);
  };`

Levi-Armstrong, 2022-05-12:
You need to create your own variable set by inheriting from VariableSet like what is shown in this [example](https://github.com/ethz-adrl/ifopt/blob/f884e80aa875de8dc07ea8fef133e5f9280c7048/ifopt_core/test/ifopt/test_vars_constr_cost.h#L62-L101). 

Are you wanting to know how to access the x0_ and x1_ from your cost or constraint function?

naguil, 2022-05-12:
Yes, I already have a variable set, and I wanted to access x0_ and x1_ (two Fourier series in my case) from the constraint set. If it isn't possible, then I fail to see the overall usefulness of x0_ and x1_

Levi-Armstrong, 2022-05-12:
Add two methods to your variable set class `getX0()` and `getX1()` then in your cost or constraint.
``` c++
auto my_set = std::dynamic_pointer_cast<MyVariableType>(GetVariables()->GetComponent("var_set1"));
... now call your methods exposing x0_ and x1_
```

naguil, 2022-05-12:
Thank you, I'll try it out.

naguil, 2022-05-13:
> Add two methods to your variable set class `getX0()` and `getX1()` then in your cost or constraint.
> 
> ```c++
> auto my_set = std::dynamic_pointer_cast<MyVariableType>(GetVariables()->GetComponent("var_set1"));
> ... now call your methods exposing x0_ and x1_
> ```

Thank you very much. It worked like magic. I hadn't heard about dynamic_pointer_cast. I was afraid the solution was something much simpler that I wasn't seeing.

Levi-Armstrong, 2022-05-13:
Yea, I think this is the only option you have if you want implementation specific access.

closed 2022-05-13 by Levi-Armstrong