ethz-adrl/ifopt #83: GetBounds() problem when both boundaries are negative

TomasMerva, 2022-06-22:
Hi, I need to add joint limits and one of the joint have both limits negative `Bounds(-3.0718, -0.0698)`. I have set the initial value of the decision variable as `-0.5` but the NLP is saying that the variable is violated. When I change the upper limit to `0`, the number of violated variables is zero. Is it not possible to use both negative limits or the mistake is elsewhere?

   ```cpp
 int time_steps = 5;
    VecBound bounds(GetRows());
    for (int i=0; i<time_steps; i++)
    {
        bounds.at(i*7+0) = Bounds(-2.8973, 2.8973);  // joint_1
        bounds.at(i*7+1) = Bounds(-1.7628, 1.7628);  // joint_2
        bounds.at(i*7+2) = Bounds(-2.8973, 2.8973);  // joint_3

        // -------This line is problematic-----------
        bounds.at(i*7+3) = Bounds(-3.0718, -0.0698); // joint_4
        // -----------------------------------------

        bounds.at(i*7+4) = Bounds(-2.8973, 2.8973);  // joint_5
        bounds.at(i*7+5) = Bounds(-0.0175, 3.7525);  // joint_6
        bounds.at(i*7+6) = Bounds(-2.8973, 2.8973);  // joint_7
    }
    return bounds;
```
[image]

Levi-Armstrong, 2022-06-22:
Can you post the compiler output error? Also if you don't mind posting it as text instead of an image this way it is searchable.

TomasMerva, 2022-06-22:
I am so sorry, I have found a mistake within the `GetValues()` method...now everything is working. Thank you for your effort and I am closing the issue.

closed 2022-06-22 by TomasMerva