ethz-adrl/ifopt #85: [Q] Weak solver performance

TomasMerva, 2022-07-11:
Hi, 
firstly I am not sure if this is the right place to ask this type of questions, so I am sorry if it is not. I have implemented the same algorithm (KOMO motion planner for Franka Emika) in Casadi (using IPOPT) and in ifopt (using IPOPT), however the required time for solving the same problem in ifopt is over 0.8 seconds where in Casadi it is around 5ms. I have tried to set same solver options but without any luck. I do not have enough experience to know where to find errors according to the IPOPT logs so could you point me to the things that could be wrong? Thank you in advance

**ifopt terminal outputs:**
```
************************************************************
    IFOPT - Interface to Nonlinear Optimizers (v2.0)
                © Alexander W. Winkler
           https://github.com/ethz-adrl/ifopt
************************************************************
Legend:
c - number of variables, constraints or cost terms
i - indices of this set in overall problem
v - number of [violated variable- or constraint-bounds] or [cost term value]

                                   c             i             v 
variable-sets:
   x                              140        0....139          18

constraint-sets:

cost-terms:
   k_order=2                        1        0......0       35.19

******************************************************************************
This program contains Ipopt, a library for large-scale nonlinear optimization.
 Ipopt is released as open source code under the Eclipse Public License (EPL).
         For more information visit http://projects.coin-or.org/Ipopt
******************************************************************************

Starting derivative checker for first derivatives.

No errors detected by derivative checker.

Total number of variables............................:      126
                     variables with only lower bounds:        0
                variables with lower and upper bounds:      126
                     variables with only upper bounds:        0
Total number of equality constraints.................:        0
Total number of inequality constraints...............:        0
        inequality constraints with only lower bounds:        0
   inequality constraints with lower and upper bounds:        0
        inequality constraints with only upper bounds:        0

Number of Iterations....: 164

                                   (scaled)                 (unscaled)
Objective...............:   1.9920397525905261e-03    1.9920397525905261e-03
Dual infeasibility......:   7.4954637053187387e-04    7.4954637053187387e-04
Constraint violation....:   0.0000000000000000e+00    0.0000000000000000e+00
Complementarity.........:   1.0286094458408624e-11    1.0286094458408624e-11
Overall NLP error.......:   7.4954637053187387e-04    7.4954637053187387e-04

Number of objective function evaluations             = 1084
Number of objective gradient evaluations             = 165
Number of equality constraint evaluations            = 0
Number of inequality constraint evaluations          = 0
Number of equality constraint Jacobian evaluations   = 0
Number of inequality constraint Jacobian evaluations = 0
Number of Lagrangian Hessian evaluations             = 0
Total CPU secs in IPOPT (w/o function evaluations)   =      0.386
Total CPU secs in NLP function evaluations           =      0.114

EXIT: Optimal Solution Found.
```

**casadi with ipopt terminal output**
```
******************************************************************************
This program contains Ipopt, a library for large-scale nonlinear optimization.
 Ipopt is released as open source code under the Eclipse Public License (EPL).
         For more information visit http://projects.coin-or.org/Ipopt
******************************************************************************

This is Ipopt version 3.12.3, running with linear solver mumps.
NOTE: Other linear solvers might be more efficient (see Ipopt documentation).

Number of nonzeros in equality constraint Jacobian...:        0
Number of nonzeros in inequality constraint Jacobian.:        0
Number of nonzeros in Lagrangian Hessian.............:      357

Total number of variables............................:      126
                     variables with only lower bounds:        0
                variables with lower and upper bounds:      126
                     variables with only upper bounds:        0
Total number of equality constraints.................:        0
Total number of inequality constraints...............:        0
        inequality constraints with only lower bounds:        0
   inequality constraints with lower and upper bounds:        0
        inequality constraints with only upper bounds:        0

iter    objective    inf_pr   inf_du lg(mu)  ||d||  lg(rg) alpha_du alpha_pr  ls
   0  2.1253940e+02 0.00e+00 1.71e+01  -1.0 0.00e+00    -  0.00e+00 0.00e+00   0
   1  2.0206655e+01 0.00e+00 6.90e+00  -1.0 1.20e+00    -  1.14e-01 1.00e+00f  1
   2  6.5670904e+00 0.00e+00 1.37e+00  -1.0 8.16e-01    -  3.74e-01 1.00e+00f  1
   3  3.2177296e+00 0.00e+00 4.20e-01  -1.0 5.31e-01    -  5.25e-01 1.00e+00f  1
   4  1.1957934e+00 0.00e+00 1.43e-01  -1.0 5.65e-01    -  6.74e-01 1.00e+00f  1
   5  3.8116699e-01 0.00e+00 3.01e-02  -1.7 4.15e-01    -  8.03e-01 1.00e+00f  1
   6  9.8561908e-02 0.00e+00 9.94e-03  -2.5 6.37e-01    -  8.29e-01 1.00e+00f  1
   7  3.3479435e-02 0.00e+00 2.59e-03  -2.5 5.73e-01    -  8.77e-01 1.00e+00f  1
   8  7.9561309e-03 0.00e+00 1.56e-03  -3.8 5.31e-01    -  7.55e-01 1.00e+00f  1
   9  3.0230167e-03 0.00e+00 5.81e-15  -3.8 3.45e-01    -  1.00e+00 1.00e+00f  1
iter    objective    inf_pr   inf_du lg(mu)  ||d||  lg(rg) alpha_du alpha_pr  ls
  10  2.5830397e-03 0.00e+00 3.10e-05  -5.7 1.39e-01    -  9.45e-01 1.00e+00f  1
  11  2.5749846e-03 0.00e+00 5.29e-15  -5.7 2.25e-02    -  1.00e+00 1.00e+00f  1
  12  2.5749615e-03 0.00e+00 7.55e-15  -8.6 1.21e-03    -  1.00e+00 1.00e+00f  1

Number of Iterations....: 12

                                   (scaled)                 (unscaled)
Objective...............:   2.5749614818928864e-03    2.5749614818928864e-03
Dual infeasibility......:   7.5478594533688798e-15    7.5478594533688798e-15
Constraint violation....:   0.0000000000000000e+00    0.0000000000000000e+00
Complementarity.........:   7.6332274818848167e-09    7.6332274818848167e-09
Overall NLP error.......:   7.6332274818848167e-09    7.6332274818848167e-09

Number of objective function evaluations             = 13
Number of objective gradient evaluations             = 13
Number of equality constraint evaluations            = 0
Number of inequality constraint evaluations          = 0
Number of equality constraint Jacobian evaluations   = 0
Number of inequality constraint Jacobian evaluations = 0
Number of Lagrangian Hessian evaluations             = 12
Total CPU secs in IPOPT (w/o function evaluations)   =      0.002
Total CPU secs in NLP function evaluations           =      0.000

EXIT: Optimal Solution Found.
      solver  :   t_proc      (avg)   t_wall      (avg)    n_eval
       nlp_f  |  36.00us (  2.77us)  33.91us (  2.61us)        13
  nlp_grad_f  |  47.00us (  3.36us)  46.08us (  3.29us)        14
  nlp_hess_l  |  43.00us (  3.58us)  42.09us (  3.51us)        12
       total  |   4.08ms (  4.08ms)   4.08ms (  4.08ms)         1
```

Levi-Armstrong, 2022-07-11:
The difference is most likely due to casadi enabling hessian and IFOPT is not.

Levi-Armstrong, 2022-07-11:
Where you create the IpoptSolver you will need to enable hessian_approximation. [Here](https://coin-or.github.io/Ipopt/OPTIONS.html#OPT_hessian_approximation) is more documentation on option. I would figure out how to print the options used by casadi to get an accurate comparison.

``` c++
ifopt::IpoptSolver ipopt;
ipopt.SetOption("hessian_approximation", "limited-memory");
```

TomasMerva, 2022-07-11:
Thank you for your reply. I have tried this option, but it seems to have almost no effect. The thing that is weird to me is that the `Number of Lagrangian Hessian evaluations` is still zero. So far I have tried to change these options:
```cpp
  ipopt.SetOption("linear_solver", "mumps");
  ipopt.SetOption("jacobian_approximation", "exact");
  ipopt.SetOption("hessian_approximation", "limited-memory");
  ipopt.SetOption("acceptable_tol", 1e-2);
```
I will try to print all options from casadi to compare.

```
Number of Iterations....: 127

                                   (scaled)                 (unscaled)
Objective...............:   1.9850253637901925e-03    1.9850253637901925e-03
Dual infeasibility......:   9.6753153678974481e-04    9.6753153678974481e-04
Constraint violation....:   0.0000000000000000e+00    0.0000000000000000e+00
Complementarity.........:   1.0000995727557652e-11    1.0000995727557652e-11
Overall NLP error.......:   9.6753153678974481e-04    9.6753153678974481e-04

Number of objective function evaluations             = 940
Number of objective gradient evaluations             = 128
Number of equality constraint evaluations            = 0
Number of inequality constraint evaluations          = 0
Number of equality constraint Jacobian evaluations   = 0
Number of inequality constraint Jacobian evaluations = 0
Number of Lagrangian Hessian evaluations             = 0
Total CPU secs in IPOPT (w/o function evaluations)   =      0.386
Total CPU secs in NLP function evaluations           =      0.123
```

Levi-Armstrong, 2022-07-11:
https://github.com/casadi/casadi/blob/master/casadi/solvers/sqpmethod.cpp

TomasMerva, 2022-07-11:
Ok, so as you suggested I believe that the problem is with using different settings for IPOPT. I am closing the issue and if I find the "best" settings, I will post it later. Thank you once more for your help.

closed 2022-07-11 by TomasMerva

Levi-Armstrong, 2022-07-11:
@TomasMerva I would be curious to see your comparison using casadi; is it somewhere public?

TomasMerva, 2022-07-12:
@Levi-Armstrong When I set `options["ipopt.hessian_approximation"] = "exact"` within Casadi, I am getting the same time as in ifopt. Thank you

Here is the link for Casadi implementation:
https://github.com/TomasMerva/OptimizationAlgorithms/blob/00a05753ffd28139a798b2781747259b929d7c0a/KOMO/komo_v1/komo_basic.py#L110